bitbank2/ss_oled #1: FONT_SIZE

GreyArea1966, 2019-07-14:
Hi.

I'm running a 128x32 display with your library. I'm sure at one point I could get the large font to occupy two lines, so there was space for two other lines in either small or normal...now it seems to take up the whole display.

What am I doing wrong?

bitbank2, 2019-07-14:
I recently made a change for AVR versus non-AVR MCUs. On AVR MCUs due to the shortage of FLASH, the 16x32 (large) font is not used and instead the 8x8 font is stretched to 16x16. On non-AVR MCUs, the original 16x32 font is used when FONT_LARGE is chosen. You can easily patch the code to do the stretching on all MCUs by looking for #ifndef/#ifdef __AVR__ in the code

GreyArea1966, 2019-07-14:
Ah, this is another "I switched from a Pro Trinket to a Trinket M0" issue then...thanks for explaining, I'll see if I can address.

Not really an issue, but related - if I find out the buffer pointer, will/could using Bitshift on that address result in the possibility of a scrolling display? If it's not that simple, I'd suggest some sort of scroll commands would be a great addition to your already excellent library.

bitbank2, 2019-07-14:
Yes, you can certainly modify the backing buffer and then use oledDumpBuffer() to write a new copy to the OLED. I'm looking at making some upgrades to my OLED and LCD libraries for more capable MCUs. At the moment, you can specify the horizontal position in pixels, so another way to accomplish scrolling is to just rewrite the line at shifting X positions and get a scrolling effect.

GreyArea1966, 2019-07-14:
I tried that, it's a bit.,..flickery.

On the other hand, I delved into your library and made the changes to make four font choices. Currently a little overwhelmed that it worked! First time!

I also changed the order...in my version I now have 0/1/2/3 as small/normal/stretched/large.

Not bad for someone who doesn't really have a clue...thanks for the guidance!

bitbank2, 2019-07-14:
glad you got it working. If your code looks reasonable, do a pull request and I'll add it

GreyArea1966, 2019-07-14:
Probably will give you a problem...I took out the #ifdef for the AVR, so it will probably break a smaller memory board.

Plus, you'd have to teach me how to do a pull request, lol.

All I really changed was this bit starting at line 1068 in ss_oled.cpp;

[code]
//
// Draw a string of small (6x8), normal (8x8), stretched (16x16) or large (16x32) characters
// At the given col+row
//
int oledWriteString(int x, int y, char *szMsg, int iSize, int bInvert)
{
int i, iFontOff;
unsigned char c, *s, ucTemp[40];

    oledSetPosition(x, y);
    if (iSize == FONT_NORMAL) // 8x8 font
    {
       i = 0;
       while (x < oled_x-7 && szMsg[i] != 0)
       {
         c = (unsigned char)szMsg[i];
         iFontOff = (int)(c-32) * 8;
         // we can't directly use the pointer to FLASH memory, so copy to a local buffer
         memcpy_P(ucTemp, &ucFont[iFontOff], 8);
         if (bInvert) InvertBytes(ucTemp, 8);
//         oledCachedWrite(ucTemp, 8);
         oledWriteDataBlock(ucTemp, 8); // write character pattern
         x += 8;
         i++;
       }
//    oledCachedFlush(); // write any remaining data
    }
    else if (iSize == FONT_STRETCHED) // 16x32 font
    {
      i = 0;
      while (x < oled_x-15 && szMsg[i] != 0)
      {
// stretch the 'normal' font instead of using the big font
          int tx, ty;
          c = szMsg[i] - 32;
          unsigned char uc1, uc2, ucMask, *pDest;
          s = (unsigned char *)&ucFont[(int)c*8];
          memcpy_P(ucTemp, s, 8);
          if (bInvert)
              InvertBytes(ucTemp, 8);
          // Stretch the font to double width + double height
          memset(&ucTemp[8], 0, 32); // write 32 new bytes
          for (tx=0; tx<8; tx++)
          {
              ucMask = 3;
              pDest = &ucTemp[8+tx*2];
              uc1 = uc2 = 0;
              c = ucTemp[tx];
              for (ty=0; ty<4; ty++)
              {
                  if (c & (1 << ty)) // a bit is set
                      uc1 |= ucMask;
                  if (c & (1 << (ty + 4)))
                      uc2 |= ucMask;
                  ucMask <<= 2;
              }
              pDest[0] = uc1;
              pDest[1] = uc1; // double width
              pDest[16] = uc2;
              pDest[17] = uc2;
          }
          oledSetPosition(x, y);
          oledWriteDataBlock(&ucTemp[8], 16);
          oledSetPosition(x, y+1);
          oledWriteDataBlock(&ucTemp[24], 16);
          x += 16;
          i++;
       }
    }

    else if (iSize == FONT_LARGE) // 16x32 font
    {
      i = 0;
      while (x < oled_x-15 && szMsg[i] != 0)
      {
          s = (unsigned char *)&ucBigFont[(unsigned char)(szMsg[i]-32)*64];
          // we can't directly use the pointer to FLASH memory, so copy to a local buffer
          oledSetPosition(x, y);
          memcpy_P(ucTemp, s, 16);
          if (bInvert) InvertBytes(ucTemp, 16);
          oledWriteDataBlock(ucTemp, 16); // write character pattern
          oledSetPosition(x, y+1);
          memcpy_P(ucTemp, s+16, 16);
          if (bInvert) InvertBytes(ucTemp, 16);
          oledWriteDataBlock(ucTemp, 16); // write character pattern
          if (y <= 5)
          {
             oledSetPosition(x, y+2);
             memcpy_P(ucTemp, s+32, 16);
             if (bInvert) InvertBytes(ucTemp, 16);
             oledWriteDataBlock(ucTemp, 16); // write character pattern
          }
          if (y <= 4)
          {
             oledSetPosition(x, y+3);
             memcpy_P(ucTemp, s+48, 16);
             if (bInvert) InvertBytes(ucTemp, 16);
             oledWriteDataBlock(ucTemp, 16); // write character pattern
          }
          x += 16;
          i++;
       }
    }
    else if (iSize == FONT_SMALL) // 6x8 font
    {
       i = 0;
       while (x < oled_x-5 && szMsg[i] != 0)
       {
         c = szMsg[i] - 32;
         // we can't directly use the pointer to FLASH memory, so copy to a local buffer
         memcpy_P(ucTemp, &ucSmallFont[(int)c*6], 6);
         if (bInvert) InvertBytes(ucTemp, 6);
         oledWriteDataBlock(ucTemp, 6); // write character pattern
//         oledCachedWrite(ucTemp, 6);
         x += 6;
         i++;
       }
//    oledCachedFlush(); // write any remaining data      
    }
  return 0;
} /* oledWriteString() */
[/code]

and these definitions in the ss_oled.h file

[code]
#define FONT_SMALL 0			// 6x8
#define FONT_NORMAL 1      	     // 8x8
#define FONT_STRETCHED 2 		// 8x16
#define FONT_LARGE 3			// 16x32
[/code]

Just curious...as a novice programmer, I'd have been tempted to use a "switch" statement on the "iSize" variable rather than all those if/thens...is there a reason you didn't?

GreyArea1966, 2019-07-14:
Eek. WTH happened to my formatting?

GreyArea1966, 2019-07-14:
Just thought, easy way to put the acr, check back in would now be as simple as reinstating the ifdef AVR and then another if statement to change iSize from FONT_LARGE to FONT_STRETCHED.

bitbank2, 2019-07-14:
It's not a problem - I'll make the change and test it across a couple of MCUs.

GreyArea1966, 2019-07-14:
Thanks...it's probably better you write it...I really, truly don't know what I'm doing...a little knowledge etc etc.

It's just I find that ability to have one line of large and two lines of small or normal really useful.

by the way...a free bit of code for you...draws a nice sine curve, scaled to fill any display;

int width = 128;
int height = 32;
void loop() {
  float i, x[2], y[2];

  oledFill(0);                                                //clear screen
  for (x[0] = 0; x[0] < 360; x[0]++) {
    y[0] = sin(x[0] * PI / 180);                       //sine requires radians
    x[1] = x[0]/360*width;                            //scales for width
    y[1] = (height/2)+(y[0] * (height/2));     //scales for height and allows for negative side of sine curve
    oledSetPixel(x[1], y[1], 1);                      //write to screen
    delay(10);                                              //pause
  }
delay(2000);                                               //hold on screen 2 secs
}

bitbank2, 2019-07-15:
Added stretch font and fixed an error in the large font data.

closed 2019-07-15 by bitbank2