bitbank2/ss_oled #9: Coordinates of SH1106 1.3" I2C are shifted after multiple calls of oledSetPixel()

vitasam, 2019-08-08:
Hi,
still there are some problems with SH1106 1.3" I2C OLED and oledSetPixel(). I modified **simple_demo_avr.ino** sketch by adding 2 functions:

```
void draw_horz_line(int x1, int x2, int y) {
    int i;
    for(i = x1; i <= x2; i++)
        oledSetPixel(i, y, 1, 1);
}

void draw_vert_line(int x, int y1, int y2) {
    int i;
    for(i = y1; i <= y2; i++)
        oledSetPixel(x, i, 1, 1);
}
```
The loop() function modified like this:

```
void loop()
{
    int i, x, y;

    oledFill(0, 1);

    draw_horz_line(10, 105, 5);
    draw_vert_line(10, 5, 45);
    draw_horz_line(10, 105, 45);

    delay(2000);
    
    oledWriteString(0, 16, 0,(char *)"ss_oled Demo", FONT_NORMAL, 0, 1);
    oledWriteString(0, 0, 1,(char *)"Written by Larry Bank", FONT_SMALL, 1, 1);
    oledWriteString(0, 0, 3,(char *)"**Demo**", FONT_STRETCHED, 0, 1);
    oledWriteString(0, 9, 6,(char *)"for AVR", FONT_STRETCHED, 0, 1);

    delay(2000);
    oledFill(0, 1);
    delay(2000);
}
```
Results: a) only 2 lines are visible; b) coordinates are shifted:
[image]
[image]

bitbank2, 2019-08-08:
I've seen this occur randomly on my displays also. I'm not sure if it's a bug in the controller or some other reason. The commands to enter and leave "read-modify-write" mode are well documented and I'm executing them correctly. I really have no idea how to fix this. You can do quite a lot of drawing operations without having a backing buffer and without using that SH1106 specific feature. All I can suggest is to avoid using it.

closed 2019-08-08 by bitbank2

vitasam, 2019-08-08:
Ok, maybe it could be even HW problem. Thanks!

bitbank2, 2019-08-08:
I don't think it's a problem with your specific display; all of mine randomly do this. It's the 'random' part which makes it hard to diagnose. I'm following the instructions in their datasheet and the bytes get written correctly most of the time, but after so many iterations, the controller "flips out". I cannot explain it :(

vitasam, 2019-08-08:
I just found this one: https://github.com/wonho-maker/Adafruit_SH1106
The whole demo works already 15 minutes with many lines and drawings - seems that it is quite stable. Although that library seems to be more fatty that your's

bitbank2, 2019-08-08:
That one doesn't appear to be using the "read-modify-write" mode of the chip (the source of trouble). If you enable the back buffer in my code, you'll have no problem doing all of the operations, but it will take 1K of your precious RAM.

vitasam, 2019-08-08:
Ah, Ok, understand!

vitasam, 2019-08-08:
Hi,
seems that there is another problem. After enabling USE_BACKBUFFER flag a following code
```
void setup()
{
    int rc;
    rc = oledInit(OLED_128x64, 0, 0, -1, -1, 400000L);     // Use standard I2C bus at 400Khz
    if (rc == OLED_NOT_FOUND)
    {
        while(1);
    }
    else
    {
        oledFill(0, 1);
        oledDrawLine(10, 15, 120, **25**, 1);
        while(1);
    }
}
```
with y2 = 25, 35, 45 returns following artifacts respectively:

[image]

[image]

[image]

bitbank2, 2019-08-09:
You're erasing the display and drawing a line in a tight loop. The artifacts you see are because the display is flashing quickly between blank and a line and the pixel refresh is not in sync with this flashing. Remove the oledFill(0,1) from the inner loop.

bitbank2, 2019-08-09:
I just tested this on an ATmega328 with a SH1106 and the latest ss_oled code. I am not able to reproduce the problem you're seeing. Mine draws the lines all correctly.
[image]

vitasam, 2019-08-09:
Ok, then it looks like a HW issue with my display. Thanks!