bitbank2/ss_oled #13: Clearing the display

elgranshaman, 2019-10-02:
Hello,

Is there a "smooth" way of clearing the display? When using oledFill(0x0, 1); it has some tearing effect that doesn't look very good when you are constantly updating the display.

Thank you

bitbank2, 2019-10-02:
oledFill is basically filling the display as fast as possible. If you're using an I2C display, the clock rate has a lot to do with how it looks while it's doing that. At 100Khz (the default speed for I2C on Arduino and Raspberry Pi), it will look slow and visible. Most SSD1306 displays are easily able to handle 1Mhz - so the faster the better. The other problem is if you're filming it, the default frame rate is 70hz (IIRC), which will have a 'beat frequency' with your camera. Then there's the problem of power supply variations. If you're flashing content between light and dark, the power supply may not keep up and you'll get a 'bloom' or 'fade' effect.

closed 2019-10-02 by bitbank2

elgranshaman, 2019-10-02:
Is there a way to increase the I2C clock frequency using the library or do I have to modify the MCU config registers directly?

bitbank2, 2019-10-02:
The clock frequency submitted in the oledInit() call will be used. On most Arduinos, the hardware I2C only allows you to set frequencies up to 400Khz. Cortex-M0's allow any speed, so you can try larger values. If you're using my bit bang code (by specifying SDA and SCL pins), I let you set higher speeds. Read the README about pin numbering if you're using bit banging on AVR.