bitbank2/ss_oled #14: [Feature request] Support RST line

cyberman54, 2019-10-04:
Can we have RST line support for SSD1306 in the library?

bitbank2, 2019-10-04:
It is already supported. See https://github.com/bitbank2/ss_oled/blob/master/src/ss_oled.cpp line 691.
If this is not sufficient, can you explain better what you're looking for?

cyberman54, 2019-10-04:
...but only for SPI interface, not for I2C:

bitbank2, 2019-10-04:
The common SSD1306 modules available to the public don't expose the reset line when they come configured for I2C. It's not exposed because it's not needed. Can you show me a situation where the reset line is needed on a SSD1306 configured for I2C?

closed 2019-10-04 by bitbank2

cyberman54, 2019-10-04:
On Heltec ESP32 Wifi Lora32 boards, v1 and v2, the SSD1306 is I2C driven AND the RST line is connected. This is useful to clear the display.

cyberman54, 2019-10-04:
Look here:

bitbank2, 2019-10-04:
One vendor selling one board which has the reset line exposed is not a compelling argument. If you want to add that functionality, then I suggestion you fork my library and add it. Otherwise it doesn't add any value to the vast majority of people using it.