bitbank2/ss_oled #15: Bug with ATTiny85

JanekLeVert, 2019-10-04:
Hi,

First of all I have to say that you're doing a fantastic job so just keep up the good work man.

That said I identified a bug in your ss_oled.cpp code int the following situation:

Context:
- OS: OSX 10.14.6
- IDE: Arduino IDE 1.8.10
- Board: ATtiny85 8 Mhz
- Programmer AVR for Attiny85 and so USBtiny (or FABtiny) depending on how you name it while installing drivers
- Libraries: BitBang_I2C and ss_oled
- Screen: 128x64 tested successfully on Arduino Mega (SCL=21, SDA=20)
- SCL=2 on ATtiny85
- SDA=0 on ATtiny85

Compiling problem:
I have an error related to SPI that is not declared in _I2CWrite when in Arduino.
There is this code: I put in bold the code that generates the compiling error.

> // Wrapper function to write I2C data
> #ifdef _LINUX_
> static void _I2CWrite(unsigned char *pData, int iLen)
> {
>   write(file_i2c, pData, iLen);
> }
> #else // Arduino
> static void _I2CWrite(unsigned char *pData, int iLen)
> {
>   if (iCSPin != -1) // we're writing to SPI, treat it differently
>   {
>     digitalWrite(iDCPin, (pData[0] == 0) ? LOW : HIGH); // data versus command
>     digitalWrite(iCSPin, LOW);
> #ifdef HAL_ESP32_HAL_H_ 
>    {
>    uint8_t ucTemp[1024];
>         SPI.transferBytes(&pData[1], ucTemp, iLen-1);
>    }
> #else
>     **SPI.transfer(&pData[1], iLen-1);**
> #endif
>     digitalWrite(iCSPin, HIGH);
>   }
>   else // must be I2C
>   {
>     I2CWrite(oled_addr, pData, iLen);
>   } // I2C
> } /* _I2CWrite() */
> #endif // _LINUX_
> 

To avoid having this compiling error I added compilation directives:

> // Wrapper function to write I2C data
> #ifdef _LINUX_
> static void _I2CWrite(unsigned char *pData, int iLen)
> {
>   write(file_i2c, pData, iLen);
> }
> #else // Arduino
> static void _I2CWrite(unsigned char *pData, int iLen)
> {
>   if (iCSPin != -1) // we're writing to SPI, treat it differently
>   {
>     digitalWrite(iDCPin, (pData[0] == 0) ? LOW : HIGH); // data versus command
>     digitalWrite(iCSPin, LOW);
> #ifdef HAL_ESP32_HAL_H_ 
>    {
>    uint8_t ucTemp[1024];
>         SPI.transferBytes(&pData[1], ucTemp, iLen-1);
>    }
> #else
> **#ifndef __AVR_ATtiny85__**
>     SPI.transfer(&pData[1], iLen-1);
> **#endif**
> #endif
>     digitalWrite(iCSPin, HIGH);
>   }
>   else // must be I2C
>   {
>     I2CWrite(oled_addr, pData, iLen);
>   } // I2C
> } /* _I2CWrite() */
> #endif // _LINUX_
> 

and now it compiles and I think there's no impact on the way it works as iCSPin should be equal to -1 **BUT it does not work. I have no display on my I2C 128x64 LCD.**

Here are pictures of the circuit:

[image]
[image]
[image]

bitbank2, 2019-10-05:
I fixed the regression bug: https://github.com/bitbank2/ss_oled/commit/1b5d9aed2beea5b064ae2fc8c7c9a7c4c2695d70

As far as not working, it's hard to diagnose from what you've shared. For ATtiny85, use the port pin numbering. e.g. instead of pin 0, use 0xb0 (PORTB, bit 0). This will use my faster direct-port access code.

closed 2019-10-05 by bitbank2

JanekLeVert, 2019-10-05:
I tested my ATtiny to verify pin 0 and 2 work well and they do.
I tested once again your code on the Arduino Mega (pin 20 and 21) and it works well.
Both code call **I2CWrite(oled_addr, pData, iLen);** but one works, the other doesn't. The difference between the 2 circuits is the microcontroller:

**Arduino:**
- SDA: pin 20
- SCL: pin 21
- VCC: 3.3V
- GND

**ATtiny85:**
- SDA: pin 0 (0xb0) <-- I don't see why 0xb0 is different to 0 once compiled
- SCL: pin 2 (0xb2) <-- "                          0xb2                        2
- VCC: upper right pin
- GND: lower left pin

I would be very interested to see screenshot of your circuit where I can pins on your attiny85 and pins one your I2C display.

Here is mine
[image]

bitbank2, 2019-10-05:
There's a big difference between specifying pin 0 and pin 0xb0 in my code (see the bitbang code which gets called to output I2C data). Share your sketch in its entirety; perhaps you're not initializing it correctly.

JanekLeVert, 2019-10-05:
Thanks for helping me solving that problem (I love your lib):

Just use your own code: simple_demo_avr.ino

```
//
// Small Simple OLED library demo for AVR platform,
// without line drawing function (saving of 1K RAM)
//
#include <ss_oled.h>

void setup()
{
    int rc;
    //rc = oledInit(OLED_128x64, 0, 0, -1, -1, 400000L);       // Standard HW I2C bus at 400Khz
    rc = oledInit(OLED_128x64, 0, 0, 0xb0, 0xb2, 400000L); // for ATtiny85, use P0 as SDA and P2 as SCL
    //rc = oledInit(OLED_128x64, 0, 0, 20, 21, 400000L); // for Arduino Mega2560

    if (rc != OLED_NOT_FOUND)
    {
        char *msgs[] =
        {
          "SSD1306 @ 0x3C",
          "SSD1306 @ 0x3D",
          "SH1106 @ 0x3C",
          "SH1106 @ 0x3D"
        };

        oledFill(0, 1);
        oledWriteString(0, 0, 0, (char *)"OLED found:", FONT_NORMAL, 0, 1);
        oledWriteString(0, 10, 2, msgs[rc], FONT_NORMAL, 0, 1);
        delay(3000);
    }
}

void loop()
{
    int i, x, y;

    oledFill(0, 1);
    oledWriteString(0, 16, 0,(char *)"ss_oled Demo", FONT_NORMAL, 0, 1);
    oledWriteString(0, 0, 1,(char *)"Written by Larry Bank", FONT_SMALL, 1, 1);
    oledWriteString(0, 0, 3,(char *)"**Demo**", FONT_STRETCHED, 0, 1);
    oledWriteString(0, 9, 6,(char *)"for AVR", FONT_STRETCHED, 0, 1);

    delay(2000);
    oledFill(0, 1);

    for (i = 0; i < 1000; i++)
    {
        x = random(128);
        y = random(64);
        oledSetPixel(x, y, 1, 1);
    }

    delay(2000);
}
```

bitbank2, 2019-10-05:
Looks like you're specifying the wrong pins. You've got SDA connected to PB0 (0) and SCL connected to PB1 (1).
[image]

JanekLeVert, 2019-10-05:
My photo was not taken from the right angle. Here is a better one. You'll see that SDA in on PB0 and SCL on PB2.

[image]

I really don't understand the problem. Could it be the frequency when plugged to ATtiny85 that I remind you I set to 8Mhz instead of 1Mhz?

JanekLeVert, 2019-10-05:
Just tested it outside of the programmer and still does not work. Photo of the circuit.

[image]

bitbank2, 2019-10-05:
You can use any GPIO pins to drive the display. The speed of the ATtiny85 should not be a problem (fast or slow will work fine). I'm not sure why you're having difficulty. Are you using the Digispark bootloader and dev environment in the Arduino IDE? That's what I tested it on.

JanekLeVert, 2019-10-05:
Dev environment is the Arduino IDE.
I'm using this programer: https://www.gotronic.fr/art-programmateur-tiny-avr-20210.htm <-- very convenient and works very well.

JanekLeVert, 2019-10-05:
WOWWWWWWW That is just incredible. Using pin 3 and 4 instead made the job as you suggested. Thank you very much @bitbank2 that does the job.
As said Wonderfull library.

Thank you very much for your support.

bitbank2, 2019-10-05:
Glad you got it working.

JanekLeVert, 2019-10-05:
Thanks to you :-D
I have one general question. I just tried to make it working on a battery (3V) by connecting GND and VCC to it and nothing happens. It was working with 3.3V with the Arduino. Are you aware of any constraint related to voltage when using ATtiny85+Oled display?

bitbank2, 2019-10-05:
If your OLED display has a 3.3v regulator connected to the Vcc pin (most do), sending 3.0v into a 3.3v regulator will result in output that is too low (or off) to make it work. I would suggest bypassing the regulator and feed the 3.0v directly into the chip.

JanekLeVert, 2019-10-05:
I've seen that the OLED screen works the same with 3.3 and 5V. Here is a picture of the backside of the OLED screen. 

[image]

I'm not an electronic expert. Is the regulator what we have between "SELECT" and "U2"? If you were me, where would you connect VCC in this backside assuming that GND can remain where it is?

JanekLeVert, 2019-10-05:
I put 2 battery in serial (3V+3V=6V) ---> still the same ;-(. Poor I am. Once a problem is solved, a new one appears lol

bitbank2, 2019-10-05:
Don't put 6V into the ATtiny85 - it's only rated up to 5.5V. Yes, I believe the 3 pin package is the regulator. You'll need to test the connections to see which pins connect to which part of the regulator (in, gnd, out). If you use one LiPo cell to power your project (3.7-4.2v), then you won't have to worry about this issue. Or use a 3.3v boost converter to convert from 1-1.5v up to 3.3v. I use these in many of my battery powered projects.

JanekLeVert, 2019-10-05:
I don't use a LiPo cell. I want to use a small battery, let's say 1x1.5V or 2x1.5V or 1x3V (CR1225 is very small for 3V).
It sounds like the solution will be a 3.3 boost converter. I need a tiny one.

bitbank2, 2019-10-05:
This is the kind I use

https://it.aliexpress.com/item/32819454513.html

JanekLeVert, 2019-10-05:
Sounds to be a good one. Just the delay that's very long when ordering in aliexpress. If I understand well: I just have to put my batteries (1.5V or 3V) to Vi, GND in the middle and I get 3.3V from Vo knowing that I continue to use GND as usual. Correct?

bitbank2, 2019-10-05:
Yes, you understood correctly. This is the wrong place to be teaching you about electricity. Please read my blog post about using batteries with Arduinos:

https://bitbanksoftware.blogspot.com/2019/06/powering-your-arduino-with-batteries.html

JanekLeVert, 2019-10-05:
Perfect. I'll go and read your blog asap. :-D

Again thank you very much for your support all the day long. I know it took a lot of your time. Be sure I appreciate.