bitbank2/ss_oled #16: ucSmallFont not used

aa2mz, 2019-11-12:
ucSmallFont not used but it's still loaded and taking up space.
00000240 t ucSmallFont

cyberman54, 2019-11-12:
@aa2mz I can't find ucSmallFont in current code, where did you find it?

bitbank2, 2019-11-12:
The compiler tries to eliminate unused data, but since the small font is used in oledWriteString(), if you use oledWriteString for other size fonts it will probably be included. This is not a bug in my library. The small font is used (see https://github.com/bitbank2/ss_oled/blob/d68ac3d14b612226f97d462ed169b9a1f172b0aa/src/ss_oled.cpp#L1567)

Your issue is incorrect; perhaps you stated your point wrong. You're free to fork the repo and remove it in your fork if you never use it.

closed 2019-11-12 by bitbank2