bitbank2/ss_oled #17: Show variables...

Snille, 2019-11-14:
Hi, I'm trying to display "variables" form an analog source on the display (SH1107).

```
--- cut ---

void loop() {
   tempvalue = analogRead(intemp);
   lightvalue = analogRead(inlight);

   oledWriteString(0, 0, 1,(char *)"Inside", FONT_STRETCHED, 0, 1);
   oledWriteString(0, 0, 5,(char *)"Temp:", FONT_NORMAL, 0, 1);
   oledWriteString(0, 0, 7,(char *)tempvalue, FONT_NORMAL, 0, 1);
   oledWriteString(0, 0, 9,(char *)"Light:", FONT_NORMAL, 0, 1);
   oledWriteString(0, 0, 11,(char *)lightvalue, FONT_NORMAL, 0, 1);

   Serial.println("Temp: ");
   Serial.println(tempvalue);
   Serial.println("Light: ");
   Serial.println(lightvalue);
   delay(1000);
}
```

I do get the "text" correctly. But how do I get the values out from the variables?
I get the values on the serial bus... So, the variables do have values. :)

What am I missing?

Thank you!

bitbank2, 2019-11-15:
I'm not sure this is the right place to ask for help like this, but here's your answer:

char szTemp[32];
sprintf(szTemp, "%d", tempvalue);
oledWriteString(0,0,6, szTemp, FONT_NORMAL, 0, 1);

closed 2019-11-15 by bitbank2

Snille, 2019-11-15:
Thank you very much! It works. :)