bitbank2/ss_oled #20: [Feature-Request] Support multiple OLEDs

ohminy, 2020-01-03:
I have used your Multi OLED library.
It is verry awesome. but there is no various functions.
Do you have any plane to add multi-OLED function into this ss_oled library?
If that, I think, it will be super awesome ^^

bitbank2, 2020-01-03:
I put a poll on Twitter to see what people think about this idea. If it's a popular idea, I'll do it:

https://twitter.com/fast_code_r_us/status/1213032370686873603

ohminy, 2020-01-03:
Did my rash request offend you? If that, I'm so sorry.
(I don't live in English area, so it's hard to understand exactly your
feeling)

I just think Multi OLED library is very handy in some simple project.
Because I don't need to use additional multiplexer.
But, it is too simple.. lite version. It's function is too limitid.
So I just thought, it'll be more awesome, if add some functions like rotate
90 degree display or auto detect I2C address some or change fonts or moving
text.. etc

bitbank2, 2020-01-03:
Don't worry, I'm not offended. Your request is reasonable. I get very little feedback from my users, so I really don't know what people need to do with the libraries. I'm genuinely curious about the opinions. I like the idea and just want to confirm its popularity with my existing users.

hpt-dev, 2020-01-14:
I would also love to see this merged with Multi_OLED as I'm working with two identical I2C 128x32 OLED screens on the same SCL with different SDA's.  I've wanted to use some faster libraries for drawing on these but need to have them as individual objects for better manipulation.  I'll continue to tinker with this and see what I can manage but will likely be light years behind what you've done thus far!  Cheers for any help.

bitbank2, 2020-01-14:
Thank you for adding your voice to this @hpt-dev. I get so little feedback that I don't know how many people are using my libraries. I'll think of a good way to add support soon. I'm also looking at adding my ss_oled library micropython somehow.

bitbank2, 2020-02-11:
I'm working with a new board this week (W600) and came up with a new idea for allowing ss_oled to work with multiple displays. It will eliminate the need for both Multi_BitBang and Multi_OLED. I'll let you know when I have something to try.

bitbank2, 2020-02-12:
I rewrote the BitBang and SS_OLED code to work better with multiple devices. It breaks the old code :(, so I need to figure out if I'm going to release a new library or break old APIs

https://youtu.be/2rTSvroW_Yw

bitbank2, 2020-03-09:
I've done an official release with the new "multi" feature. Now you can connect an unlimited number of displays, each having their own unique characteristics.

closed 2020-03-09 by bitbank2