bitbank2/ss_oled #25: Use provided OLED address

nuthub, 2020-03-17:
When passing `iAddr` to `int oledInit(...)`, it is not passed to `pOLED->oled_addr`.

nuthub, 2020-03-17:
Thanks for merging the PR.

Please also have a look at the line [768](https://github.com/bitbank2/ss_oled/blob/729d4f7510323209bbb680234fddaf09f4511709/src/ss_oled.cpp#L768), which does not look correct to me:
```
pOLED->oled_addr = (uint8_t)scl;
```

bitbank2, 2020-03-17:
Ugly re-use of a variable specifically for Linux. Unfortunately it is correct.

nuthub, 2020-03-17:
SCL feeding into oled_addr? Shouldn't it be
```
pOLED->oled_addr = (uint8_t)iAddr;
```
?

bitbank2, 2020-03-17:
It is both wrong and right at the same time :( 
With my latest changes to the API, it needs to be cleaned up. I'm stuck away from my home (for maybe months now) and don't have an RPI handy to test it. Unless you specifically need it, it will need to stay the way it is until I can rewrite and retest it.

closed 2020-03-17 by bitbank2

nuthub, 2020-03-17:
ok, I was just wondering, if it correct, because it looked wrong. I just wanted to inform you.
Btw, thank you very much for your efforts on that library.