bitbank2/ss_oled #30: Cant compile none of examples for arduino mini pro

Rolandas3, 2020-03-29:
Hi,

i cannot compile any of examples provided for none of arduino boards. Trying to figure out i have used example "oled animation demo" and arduino mini pro board as per settings provided in exampe file

here is error :

> Arduino: 1.8.12 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp: In function 'int i2cByteOutAVR(uint8_t)':
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:294:19: error: 'iDelay' was not declared in this scope
> 
>           sleep_us(iDelay);
> 
>                    ^~~~~~
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:294:19: note: suggested alternative: 'delay'
> 
>           sleep_us(iDelay);
> 
>                    ^~~~~~
> 
>                    delay
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp: In function 'int i2cByteOutAVRFast(uint8_t)':
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:330:21: error: 'iDelay' was not declared in this scope
> 
>             sleep_us(iDelay);
> 
>                      ^~~~~~
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:330:21: note: suggested alternative: 'delay'
> 
>             sleep_us(iDelay);
> 
>                      ^~~~~~
> 
>                      delay
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:337:19: error: 'iDelay' was not declared in this scope
> 
>           sleep_us(iDelay);
> 
>                    ^~~~~~
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:337:19: note: suggested alternative: 'delay'
> 
>           sleep_us(iDelay);
> 
>                    ^~~~~~
> 
>                    delay
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:344:12: error: 'iDelay' was not declared in this scope
> 
>    sleep_us(iDelay); // DEBUG - delay/2
> 
>             ^~~~~~
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:344:12: note: suggested alternative: 'delay'
> 
>    sleep_us(iDelay); // DEBUG - delay/2
> 
>             ^~~~~~
> 
>             delay
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp: In function 'int i2cWrite(BBI2C*, uint8_t*, int)':
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:461:10: error: 'iSDA' was not declared in this scope
> 
>       if (iSDA >= 0xa0)
> 
>           ^~~~
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:461:10: note: suggested alternative: 'SDA'
> 
>       if (iSDA >= 0xa0)
> 
>           ^~~~
> 
>           SDA
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp: In function 'void I2CInit(BBI2C*, uint32_t)':
> 
> C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:504:44: error: no matching function for call to 'TwoWire::begin(uint8_t&, uint8_t&)'
> 
>            Wire.begin(pI2C->iSDA, pI2C->iSCL);
> 
>                                             ^
> 
> In file included from C:\Users\rolan\Documents\Arduino\libraries\BitBang_I2C\src\BitBang_I2C.cpp:23:0:
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:52:10: note: candidate: void TwoWire::begin()
> 
>      void begin();
> 
>           ^~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:52:10: note:   candidate expects 0 arguments, 2 provided
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:53:10: note: candidate: void TwoWire::begin(uint8_t)
> 
>      void begin(uint8_t);
> 
>           ^~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:53:10: note:   candidate expects 1 argument, 2 provided
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:54:10: note: candidate: void TwoWire::begin(int)
> 
>      void begin(int);
> 
>           ^~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:54:10: note:   candidate expects 1 argument, 2 provided
> 
> exit status 1
> Error compiling for board Arduino Pro or Pro Mini.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

program code attached:

```
#define SDA_PIN 0xb0
#define SCL_PIN 0xb1

// These pin numbers are for the M5Stack Atom
//#define SDA_PIN 21
//#define SCL_PIN 22
// Set this to -1 to disable or the GPIO pin number connected to the reset
// line of your display if it requires an external reset
#define RESET_PIN -1
// let ss_oled figure out the display address
#define OLED_ADDR -1
// don't rotate the display
#define FLIP180 0
// don't invert the display
#define INVERT 0
// Bit-Bang the I2C bus
// Set this to 1 to use the wire library
#define USE_HW_I2C 0
```

arduino version 1.8.12

Any suggestions? Big Thanks!

Br, Rolandas

bitbank2, 2020-03-29:
Which version of the library are you using? I just did some fixes and they may not have made it into the library manager yet. The latest version is 4.1.2; if you need it and it's not in the library manager, then clone it from here.

closed 2020-03-29 by bitbank2

Rolandas3, 2020-03-30:
Hi,
my library versions - 4.1.2 for ss_oled and 2.0.1 for Binbang_I2C. The issue is present
Br, Rolandas

bitbank2, 2020-03-30:
Found the problem - I had done a fix to the BitBang_I2C library, but forgot to do a new release. I just upped the release number to 2.0.2. If you can't get it quickly through the library manager, clone it directly.

Rolandas3, 2020-03-30:
Hi

Big thanks, its working now. One note :) to add I have SH1106 I2C from Aliexpress with address 0x78, but your library report SH1106 being found with 0x3c address.

message from "I2C_detector":

> Starting I2C Scan
> Device found at 0x3C, type = Unknown
> 1 device(s) found

while "SS_oled" "simple_demo" recognizes as SH1106. And when I test with 0x78 it doesnt work :). But its minor thing

Kind regards, Rolandas

bitbank2, 2020-03-30:
I2C addresses are 7-bits and an extra bit is added for READ or WRITE. Most people in "the industry" use the 7-bit form of the address when specifying it. Therefore 0x78 is really 0x3c shifted left 1 bit (two ways of describing the same thing).

Rolandas3, 2020-03-30:
Hi Laurence (hope its uour real name :) )

Ok, did not know this ;). Big thanks for a great library, will try to add this to Atmel Studio, hopefully this will work :)
Rolandas