bitbank2/ss_oled #32: [feature request] add control characters to

cyberman54, 2020-04-03:
I want to use control characters for simple print positioning.
Can we enhance `oledWriteString()` to parse (at least) some simple [control characters](https://en.wikipedia.org/wiki/Control_character) like `CR`, `LF` and `BS` ?

bitbank2, 2020-04-03:
It's a reasonable idea, but I don't want to go down that path. Writing your own string pre-processor which in turn calls oledWriteString() would be a good way to do it.

closed 2020-04-03 by bitbank2

cyberman54, 2020-04-03:
But i can't use the lib internal cursor control then, so that makes no sense.

bitbank2, 2020-04-03:
The cursor position is exposed with oledSetCursor() and each time you use oledWriteString you can either continue from the current position or specify a new one. The control is 100% in your hands.

cyberman54, 2020-04-03:
Yes, that's true, but only to print a newline i need to build my own cursor tracking.