bitbank2/ss_oled #40: How to initialise OLED with SPI

Eternal-student, 2020-08-16:
Hi Larry,
I have a project that was using the Adafruit library to display some simple text and was taking ~40ms to update an I2C OLED display.  Searching the internet I found that SPI interfaces are able to refresh quicker but also your optimised library which helped a lot with the speed of I2C.  So, I bought an SPI display (configured for 4SPI) which now updates much quicker (5ms with basic sketch) and I wondered about trying your library with the SPI display but I'm having difficulty working out how to Initialise the display.
I can see that to access the oledSPIInit routine I need to define the FUTURE variable.. does this mean it's more of an experimental feature?  I tried using a compile time define for it, but I get errors in the BitBang routine then (although maybe this isn't needed for ss_oled SPI initialising)?

This was the routine I was trying to use;
oledSPIInit(MY_OLED, 9, 10, 8, FLIP180, INVERT, 10000000L);

Can you tell me if I'm going in the wrong direction?
Do you think your library using SPI would be able to achieve faster updates than the u8x8 text only library because of the optimisation?
Do you think your I2C library can achieve similar update speeds as an SPI display?

bitbank2, 2020-08-16:
My latest OLED code is in my OneBitDisplay library and it supports SPI. It will drive the display faster than the Adafruit library. It's very similar to ss_oled, but has a lot more functionality and supports a bunch of different displays.

closed 2020-08-16 by bitbank2

Eternal-student, 2020-08-18:
Ok great, thanks!  I have SPI working with the OneBitDisplay now.. many thanks for your help and libraries!