bitbank2/ss_oled #45: How to use oledScrollBuffer()?

ArnieO, 2021-03-02:
I want to write text lines until I reach the bottom line, then scroll the entire screen upwards one line and write on the bottom line. 
I assume this is what oledScrollBuffer() is for, but after many attempts I give up figuring out how it is used. 

Below code is based on the example in ss_oled_test.ino but I see no effect on my display. 

So... anyone?
(128x64 display, so 8 lines with font 8x8)
```
void loop() {
  char buf[20];
  oledWriteString(&oled, 0, 0, line, itoa(line, buf, 10), FONT_8x8, 0, 1);
  delay(500);
  if (line == 7)
  {
    for (int i = 0; i < 8; i++) // smooth scroll 8 lines
    {
      oledScrollBuffer(&oled, 0, 127, 0, 7, 1);
      oledDumpBuffer(&oled, NULL);
      delay(40);
    }
  }
  else
    line++;
}
```

bitbank2, 2021-03-03:
It looks correct; did you define a backbuffer first with oledSetBackBuffer()?

ArnieO, 2021-03-03:
No, I did not know that was necessary. This will be implemented on an ATtiny, how much memory will that take?

bitbank2, 2021-03-03:
Unfortunately it will require 1K (128x64). There's no way around it if you want to scroll the display contents vertically 1 line at a time. If you don't need to scroll the text vertically then you don't need the display buffer.

closed 2021-03-03 by bitbank2

ArnieO, 2021-03-04:
If "smooth scrolling" (one pixel line at a time) had been available in the library at a low resource cost, I would have used it. 
But strictly speaking I just need to scroll one text line at a time - so I guess I'll just make a character buffer and manage it in my code. 
Thank you for your follow-up!

cyberman54, 2021-03-04:
If you need an example for scrolling, you may look here:

https://github.com/lenaschimmel/ESP32-Paxcounter/blob/master/src/display.cpp#L693

bitbank2, 2021-03-04:
He's using my OneBitDisplay library. I've worked with him before. It looks like he's patched it to change the OLED memory layout to be in vertical mode. Not all OLEDs support that mode. It's another way of doing it. There's a few perf issues with his code too.