bitbank2/ss_oled #46: brightness setting

80stech, 2021-03-06:
calling setcontrast only changes brightness by a very small amount even set to 0 ??

bitbank2, 2021-03-06:
Some OLED displays (e.g. 72x40) don't respond to the setcontrast command. Even though they all use the same SSD1306 controller, somehow they don't all work the same way.

closed 2021-03-06 by bitbank2

80stech, 2021-03-06:
OK thanks! :)  I have noticed that it seems to work differently on different displays I have tried. Good to know that it wasn't something I was doing wrong. Some displays will dim a fair bit and I ended up powering down the display when inactive which is probably a better way to go anyway. 

Cheers,
Rainer S.

datacute, 2021-03-07:
Most of the screens I've used have supported the extra functions described in the SSD1306B, one of which in particular: changing the current reference to be the internal current reference, I've found results in much less differences between various screen brightnesses. The 72x40 screens I've seen needed this setting as otherwise they were extremely dark as they were not supplied with an external current reference circuit.
0xAD 0x00 = External IREF
0xAD 0x10 = Internal IREF maximum ISEG 160uA
0xAD 0x30 = Internal IREF maximum ISEG 240uA