bitbank2/ss_oled #47: Usage of oledDrawSprite

d-atlas, 2021-03-08:
Hello,

How does one use the oledDrawSprite function correctly? I have been trying different byte codes generated from bmp and xbm monochrome image files and your image_to_c program (also  with 1-bpp bmp files) but I can't seem to get a picture onto my 128x64 px screen. I assume cx and cy denote the image size in pixels and x and y the location that they will be displayed to. I have defined and set a backbuffer of size 1024 on my ESP8266. Everything else works very smooth, thank you for making this library.

This is an example image I tried to display, the display was initialized at 1MHz.

```C++
#define Untitled_width 45
#define Untitled_height 45
static unsigned char Untitled_bits[] = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x02, 0x06, 0x00, 0x00, 0x00, 0x00, 0x02, 0x0e, 0x00, 0x00,
   0x00, 0x00, 0x02, 0x0a, 0x00, 0x00, 0x00, 0x00, 0x02, 0x0a, 0x00, 0x00,
   0x00, 0x00, 0x02, 0x1b, 0x00, 0x00, 0x00, 0x00, 0x02, 0x11, 0x00, 0x00,
   0x00, 0x00, 0x02, 0x11, 0x00, 0x00, 0x00, 0x00, 0x82, 0xf1, 0x00, 0x00,
   0x00, 0x00, 0x82, 0x3f, 0x00, 0x00, 0x00, 0x00, 0x82, 0x60, 0x00, 0x00,
   0x00, 0x00, 0x82, 0x40, 0x00, 0x00, 0x00, 0x00, 0x82, 0x40, 0x00, 0x00,
   0x00, 0x00, 0x06, 0x00, 0x00, 0x00, 0x00, 0x01, 0x04, 0x00, 0x00, 0x00,
   0x00, 0x01, 0xfc, 0x0f, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf8, 0x01, 0xff, 0x07, 0x00,
   0x00, 0x00, 0xff, 0x01, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x18, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

uint8_t *imgptr = (uint8_t*) &Untitled_bits[0];
oledDrawSprite(&ssoled, imgptr, 45, 45, 0, 1, 1, 0);
oledDumpBuffer(&ssoled, NULL);
```

bitbank2, 2021-03-09:
You're passing a value of 0 for the pitch. Pitch is the bytes per line of your image. You need to pass 6 (45 + 7)/8. Also you're passing the priority color as 0. I think you want to pass 1.
Have you defined a back buffer to draw it into?

d-atlas, 2021-03-09:
I didn't know that pitch was the number of bytes per line, thank you, it works now!

closed 2021-03-09 by d-atlas