bitbank2/ss_oled #48: how to install the lib on the Raspberry Pi? (Raspi OS Stretch)

dsyleixa, 2021-04-18:
hi,
how to install the lib on the Raspberry Pi? (Raspi OS Stretch)

bitbank2, 2021-04-18:
In the linux folder is a makefile which will build a standard static library. You'll link it to your code by adding -lss_oled to your linker flags.

closed 2021-04-18 by bitbank2

dsyleixa, 2021-04-18:
I tried that but it failed:

```
pi@PiStr:~/ss_oled/linux $ make
make: Warnung: Datei „makefile“ hat hat in der Zukunft liegende Änderungszeit 6835
cc -c -Wall -O2 -I../src -D_LINUX_ ss_oled.c
In file included from ../src/ss_oled.cpp:47:0,
                 from ss_oled.c:4:
../src/ss_oled.h:4:25: fatal error: BitBang_I2C.h: Datei oder Verzeichnis nicht gefunden
 #include <BitBang_I2C.h>
                         ^
compilation terminated.
makefile:12: die Regel für Ziel „ss_oled.o“ scheiterte
make: *** [ss_oled.o] Fehler 1
```

bitbank2, 2021-04-18:
You need to build BitBang_I2C first; it will install the include file so that it can be found by ss_oled.

dsyleixa, 2021-04-18:
How shall I do "build BitBang_I2C first;" ?

bitbank2, 2021-04-18:
git clone https://github.com/bitbank2/BitBang_I2C
cd BitBang_I2C/linux
make

dsyleixa, 2021-04-18:
aah, the other lib in the other repo...

ok, there was just 1 issue then about the clock (mine is actually correct though):

```
pi@PiStr:~/BitBang_I2C/linux $ make
make: Warnung: Datei „makefile“ hat hat in der Zukunft liegende Änderungszeit 7015
cc -c -Wall -O2 -I../src -D_LINUX_ BitBang_I2C.c
ar -rc libBitBang_I2C.a BitBang_I2C.o ;\
sudo cp libBitBang_I2C.a /usr/local/lib ;\
sudo cp ../src/BitBang_I2C.h /usr/local/include
make: Warnung: Mit der Uhr stimmt etwas nicht. 
         Der Bauauftrag könnte unvollständig sein.
```

dsyleixa, 2021-04-18:
then again maker in the ss_oled folder: same date issue:
```
pi@PiStr:~/ss_oled/linux $ make
make: Warnung: Datei „makefile“ hat hat in der Zukunft liegende Änderungszeit 4578
cc -c -Wall -O2 -I../src -D_LINUX_ ss_oled.c
ar -rc libss_oled.a ss_oled.o ;\
sudo cp libss_oled.a /usr/local/lib ;\
sudo cp ../src/ss_oled.h /usr/local/include
make: Warnung: Mit der Uhr stimmt etwas nicht. 
         Der Bauauftrag könnte unvollständig sein.
```

bitbank2, 2021-04-18:
I'm sorry, but this isn't a bug in the library or makefile. Github issues are not meant for tutoring people on how to use Linux.

I don't know German and I'm not sure what that error is about.

dsyleixa, 2021-04-18:
it says that the makefile has a wrong date, but then it's probably a mistake in the github repo.

so update:
into which directory do I have to copy ss_oled.h and the entire ss_oled folder that the lnker finds it by
#include <ss_oled.h>
g++ -o -lssoled
?
perhaps /usr/include? or /usr/include/libraries?

dsyleixa, 2021-04-18:
if I start compilation in the linux folder it self in /home/pi then it fails:
```
g++ -Wall -I/opt/vc/include -o  "sample" "sample.c" -pthread -I/opt/vc/include -lshapes -lwiringPi -lpigpio -lrt $(pkg-config eigen3 --cflags) -lreadline  (im Verzeichnis: /home/pi/ss_oled/linux)
sample.c: In function ‘int main(int, char**)’:
sample.c:35:68: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   oledWriteString(&ssoled, 0,0,1,"SS_OLED Library!",FONT_NORMAL,0,1);
                                                                    ^
sample.c:36:55: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   oledWriteString(&ssoled, 0,3,2,"BIG!",FONT_LARGE,0,1);
                                                       ^
sample.c:37:57: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   oledWriteString(&ssoled, 0,0,5,"Small", FONT_SMALL,0,1);
                                                         ^
/tmp/ccwDHDOU.o: In function `main':
sample.c:(.text+0xa8): undefined reference to `oledInit(ssoleds*, int, int, int, int, int, int, int, int, int)'
sample.c:(.text+0xe8): undefined reference to `oledSetBackBuffer(ssoleds*, unsigned char*)'
sample.c:(.text+0xf8): undefined reference to `oledFill(ssoleds*, unsigned char, int)'
sample.c:(.text+0x13c): undefined reference to `oledWriteString(ssoleds*, int, int, int, char*, int, int, int)'
sample.c:(.text+0x170): undefined reference to `oledWriteString(ssoleds*, int, int, int, char*, int, int, int)'
sample.c:(.text+0x1a4): undefined reference to `oledWriteString(ssoleds*, int, int, int, char*, int, int, int)'
sample.c:(.text+0x1d8): undefined reference to `oledWriteString(ssoleds*, int, int, int, char*, int, int, int)'
sample.c:(.text+0x20c): undefined reference to `oledSetPixel(ssoleds*, int, int, unsigned char, int)'
sample.c:(.text+0x230): undefined reference to `oledSetPixel(ssoleds*, int, int, unsigned char, int)'
sample.c:(.text+0x258): undefined reference to `oledPower(ssoleds*, unsigned char)'
collect2: error: ld returned 1 exit status
Kompilierung fehlgeschlagen.
```

bitbank2, 2021-04-18:
That error makes no sense. There's not an error in the repo. You need to run the makefile which will copy the library (.a) and (.h) files into the proper linux directories to be found by other projects. You do not need to manually copy anything.
1) make BitBang_I2C
2) make ss_oled
3) make your project

dsyleixa, 2021-04-18:
make BitBang_I2C <== did that
    make ss_oled <== did that

    make your project <== compiled my file by the command line:
g++ -Wall -I/opt/vc/include -o  "sample" "sample.c" -pthread -I/opt/vc/include -lshapes -lwiringPi -lpigpio -lrt $(pkg-config eigen3 --cflags) -lreadline -lss_oled

bitbank2, 2021-04-18:
This is definitely the wrong place to be chatting back and forth. contact me on Twitter Direct Messaging (@fast_code_r_us)

dsyleixa, 2021-04-18:
I don't have a twitter account

dsyleixa, 2021-04-18:
perhaps into which directory do I have to copy ss_oled.h and the entire ss_oled folder that the lnker finds it by
#include <ss_oled.h>
g++ -o -lss_oled
?
perhaps /usr/include? or /usr/include/libraries?

bitbank2, 2021-04-18:
Have you successfully built the simple examples (C) using the provided makefile?

dsyleixa, 2021-04-18:
I have typed "make" in the linux folder, but there are no executables, just
libss_loed.h
makefile
make_samples
multi_oled.h
sample.c
sample.o
speed_test.c
ss_oled.c
ss_oled.o

bitbank2, 2021-04-18:
make -f make_samples
You need to look a little better

dsyleixa, 2021-04-18:
sorry, I never use make by myself...

so, update:
```

pi@PiStr:~/ss_oled/linux $ make -f make_samples
make: Warnung: Datei „make_samples“ hat hat in der Zukunft liegende Änderungszeit 2849
cc -c -Wall -O2 -D_LINUX_ sample.c
cc sample.o -lss_oled -lpigpio -lBitBang_I2C -lm -lpthread -o sample
cc -c -Wall -O2 -D_LINUX_ multi_oled.c
cc multi_oled.o -lss_oled -lpigpio -lBitBang_I2C -lm -lpthread -o multi
cc -c -Wall -O2 -D_LINUX_ speed_test.c
cc speed_test.o -lss_oled -lpigpio -lBitBang_I2C -lm -lpthread -o speed
make: Warnung: Mit der Uhr stimmt etwas nicht. 
         Der Bauauftrag könnte unvollständig sein.
pi@PiStr:~/ss_oled/linux $ ./sample
2021-04-18 19:55:40 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

pigpio failed to initialize
2021-04-18 19:55:40 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

pigpio failed to initialize
2021-04-18 19:55:40 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

pigpio failed to initialize
Unable to initialize I2C bus 0-2, please check your connections and verify the device address by typing 'i2cdetect -y <channel>
```

dsyleixa, 2021-04-18:
ah, ok sudo needed...
now it shows the "hello" screen!
Thank you, looking forward!

dsyleixa, 2021-04-18:
...but still failes when trying to compile just from the terminal by command line
g++ -o -lss_oled .....
command line

bitbank2, 2021-04-18:
My library is C code, not C++. The function names get 'mangled' by C++ compared to C, that's why it won't link. You need to find the solution to that problem. Github issues is not a tutoring service.

dsyleixa, 2021-04-18:
ok, I need that oled lib also for my g++ projects to be included.
So I'm afraid I have to abandon that, unfortunately.

bitbank2, 2021-04-18:
You really do need a lot of help. You can easily link ss_oled (as C code) with your C++ project. All you need to do is add 'extern "C"' to the function names. You need to read a bit more on C/C++ programming.

dsyleixa, 2021-04-18:
well, as stated, I'm a beginner, and I do not know all that stuff. I just need a lib to be #included into my .c and .cpp program files (like e.g. wiringPi or pigpio , which are also plain C but can be easily linked also into C++ programs), but thank you very much.

bitbank2, 2021-04-18:
As I stated repeatedly. Your lack of experience in programming is not a valid reason to spam my github issues. You opened an issue for a problem which does not exist - I closed it. Then you opened another issue for a problem which does not exist - I closed it. Now you want me to teach you how to program. This is the problem with your activities here, not the fact that you need help.

dsyleixa, 2021-04-18:
no, I don't want to teach me anything, and I am not spamming.
My attempts for installing failed because the instructions were unclear to me, and now as it turns out that I can't use the lib I have to abandon it.
Sorry for the inconvenience.

bitbank2, 2021-04-18:
You don't have to abandon it. As I said, you just need to learn a bit more about C/C++ interfacing. It's an extremely simple problem to fix. Also, this library is not well documented because I abandoned it a long time ago. Current efforts for controlling 1-bit displays have moved to my OneBitDisplay library. It has a very detailed Wiki and lots of examples/instructions.

dsyleixa, 2021-04-18:
ok, perhaps I once  I will learn how to do this C to C++ conversion, but for the moment that's far beyond my skills  and till yet I didn't need to do thatby my own. Perhaps one day in the future (which probably will not be called "near") I'll try it again then, thank you, and again, sorry for disturbing youce .  ;-)