bitbank2/ss_oled #51: Support for 64x48 display

fijam, 2021-09-02:
Would it be possible to add support for a 1306-based 64x48px display?

Product pictured below, tested with adafruit_ssd1306 python library.

[image]

bitbank2, 2021-09-02:
I don't have that size handy to test. It isn't hard to add support, but you need to verify the init sequence (2 possible ones) and the memory offset of the visible portion of the display. At the moment I can't do that verification without having a display in hand.

closed 2021-09-02 by bitbank2

fijam, 2021-09-02:
Can you describe in more detail how to calculate the memory offset? I could give it a shot.

bitbank2, 2021-09-02:
If you look at the code, you'll see the various x/y offset values that are set up in init() depending on the display chosen. It's not a matter of calculating it, but a matter of figuring out the value by trial and error. The info is not documented and can't be guessed.