bitbank2/ss_oled #58: doesn't work for Arduino Nano Every

80stech, 2022-05-06:
I have been using this with Nanos and has been working great :) but I can't seem to get it working for the Nano Every?

bitbank2, 2022-05-06:
I have a Nano Every to test with, but I consider this library deprecated in favor of my OneBitDisplay library. Can you share the code you're using and the "mode of failure" you're seeing?

80stech, 2022-05-06:
OK, I found the problem was that I had the I2c rate cranked up to 80,0000 which has been perfectly fine using the Nanos but apparently not for the Every. I put the I2c data rate down to 40,0000 and things came to life! :) Maybe I'll try adding some pull up resistors to see if I can get the Every back up to 80,000. Thanks so much for your reply and I'll be sure to give the OneBitDisplay a try! :)

closed 2022-05-06 by bitbank2