bitbank2/ss_oled #60: Orange Pi Zero Support

f0086, 2022-09-19:
Compiling the example on ss_oled for linux on a Orange Pi Zero (Armbian) fails. To reproduce it:

```
git clone https://github.com/bitbank2/ss_oled
cd linux
make
make -f make_samples
```

Results in:

```
[...]
/usr/bin/ld: /usr/local/lib/libBitBang_I2C.a(BitBang_I2C.o): in function `I2CRead':
BitBang_I2C.c:(.text+0xa6c): undefined reference to `gpioSetMode'
/usr/bin/ld: BitBang_I2C.c:(.text+0xa80): undefined reference to `gpioSetMode'
/usr/bin/ld: BitBang_I2C.c:(.text+0xa9c): undefined reference to `gpioSetMode'
/usr/bin/ld: BitBang_I2C.c:(.text+0xab8): undefined reference to `gpioSetMode'
/usr/bin/ld: BitBang_I2C.c:(.text+0xac0): undefined reference to `gpioSetMode'
/usr/bin/ld: BitBang_I2C.c:(.text+0xad0): undefined reference to `gpioDelay'
[...]
```

BitBang_I2C (and the pigpio dependency) is installed like this:

```
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install
```
```
git clone https://github.com/bitbank2/BitBang_I2C.git
cd  linux
make
```

Not sure what  causes the error on compiling the ss_oled examples.
The OLED Display is in place and works with ```oled_96``` over i2c, channel 0.

```
orangepizero:~:% sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

bitbank2, 2022-09-20:
looks like you just need to add a -L option in the makefile for the linker to find where pigpio was installed

closed 2022-09-20 by bitbank2

f0086, 2022-09-20:
No, the problem was the order in which the libs are added in the makefile, so the linker does not drop unknown fields and functions. This is the right order:

```
LIBS = -lss_oled -lBitBang_I2C -lpigpio -lm -lpthread
```

But the core problem remains: ss_oled does not work on other Linux devices than an RaspberryPi. I've dig through both libs (BitBang_I2C and pigpio) and find no easy way to make this just use ```/dev/i2c-0``` like the abandon lib ```oled_96``` does. Here the output for the (slightly changed) example:

```
orangepizero:linux:% cat make_samples
CFLAGS=-c -Wall -O2 -D_LINUX_
LIBS = -lss_oled -lBitBang_I2C -lpigpio -lm -lpthread

all: sample

sample: sample.o
	$(CC) -L/usr/lib sample.o $(LIBS) -o sample

sample.o: sample.c
	$(CC) $(CFLAGS) sample.c

clean:
	rm *.o sample
orangepizero:linux:% cat sample.c
#include <stdint.h>
#include <string.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <ss_oled.h>

SSOLED ssoled;
unsigned char ucBackBuf[1024];

int main(int argc, char *argv[]) {
	int iChannel = 0;
	int iOLEDAddr = 0x3c;
	int iOLEDType = OLED_128x64;

	if (oledInit(&ssoled, iOLEDType, iOLEDAddr, 1, 0, 1, iChannel, iOLEDAddr, -1, 400000) != OLED_NOT_FOUND) {
		printf("Successfully opened I2C bus");
		oledSetBackBuffer(&ssoled, ucBackBuf);
		oledFill(&ssoled, 0, 1);
		oledWriteString(&ssoled, 0, 3, 2, "Hi!", FONT_LARGE, 0, 1);
		printf("Press ENTER to quit\n");
		getchar();
		oledPower(&ssoled, 0);
	} else {
		printf("Init failed!");
	}
	return 0;
}
orangepizero:linux:% make -f make_samples sample
cc -c -Wall -O2 -D_LINUX_ sample.c
cc -L/usr/lib sample.o -lss_oled -lBitBang_I2C -lpigpio -lm -lpthread -o sample
orangepizero:linux:% sudo ./sample
2022-09-20 22:41:47 gpioHardwareRevision: unknown revision=0
2022-09-20 22:41:47 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, this system does not appear to be a raspberry pi. |
|aborting.                                                |
+---------------------------------------------------------+

pigpio failed to initialize
Init failed!%                                                                                                                                                                        orangepizero:linux:%   
```

bitbank2, 2022-09-20:
I haven't used non-RPI Linux machines in a while, but I thought I had written replacement functions for the pigpio ones since it doesn't really exist on non-RPI OSs. I can't help you from this point forward. It shouldn't be hard to write the GPIO functions that are missing.